MrYsLab/telemetrix-aio #1: Not getting values after reconnecting

mklomp, 2020-11-10:
Installed TelemterixAIO via pip, and Telemetrix4Arduino via arduino library manager.

TelemetrixAIO Version: 0.02
Telemetrix4Arduino Version Number: 0.6

When running teh analog read example it works fine for the first time. But if I kill the python process and start it again it will reconnect, but no values are coming through. When I physically reset the MCU (STM32 in my case) it will work again.

MrYsLab, 2020-11-10:
Thanks. I can verify I am seeing the same thing on the stm32. The behavior is different on the Uno, so I suspect there is something with the sketch and not the client. I will investigate. If it turns out it is the sketch, I will move this issue over.
I am able to run the loop backtest when the board is in this state, so at least I should be able to debug things.

MrYsLab, 2020-11-11:
I have been trying to characterize the cause of the issue, and I have not been successful. Again, this behavior is only on the STM32 and not on an Arduino. This same behavior occurs for digital inputs as well. Here is what I have done:

1. In the example sketch I changed
```
board = telemetrix.Telemetrix()

try:
    digital_in(board, DIGITAL_PIN)
except KeyboardInterrupt:
    board.shutdown()
    sys.exit(0)

```
To:
```
board = telemetrix.Telemetrix(shutdown_on_exception=False)
board.shutdown()
time.sleep(5)
board= telemetrix.Telemetrix(shutdown_on_exception=False)

try:
    digital_in(board, DIGITAL_PIN)
except KeyboardInterrupt:
    # board.shutdown()
    # sys.exit(0)
    pass
```
I thought perhaps it was my shutdown code causing the issue.  In the second scenario, the pin never becomes functional, but the sketch is still alive (I ran the loopback example). In that scenario, on hitting control C the program simply exits. 

I also modifed the code in the while loop to be:
```
    try:
        while True:
            time.sleep(.0001)
    except KeyboardInterrupt:
        # board.shutdown()
        # sys.exit(0)
        pass
```
So my current thinking is that when the program exits, something (perhaps the serial control lines) is toggling something in the chip and puts it into a strange state. Perhaps the device is going into sleep or standby mode. I am not sure about how to prove this is true or not.

If you have any thoughts about this, please let me know. If not, the remedy seems to be having to reset the STM.

mklomp, 2020-11-12:
I think I do not completel understand why you exclude the shutdown code. I would say thay teh second scenario does indicate that there might be a problem there. But I might misunderstand what you are trying to do there.

I only ran into this issue after the first release. To my knowlegde I did not have this problem on the pre-releases. I think the last thing I checked was with phase5.

MrYsLab, 2020-11-15:
Sorry, I did not explain why I was looking at shutdown. My thinking was that the only difference between a power-up start and a restart was that the shutdown method is called. I started eliminating parts of the shutdown method to see if there was something causing the issue. No such luck. 
Thanks for the tip about phase 5. I will check things out and report back when I have some determination. Thanks for taking the time for looking at this.

MrYsLab, 2020-11-15:
I now understand the cause of the problem. I had added a function to stop all reporting, analog, digital, sonar and dht. When an Arduino brand board is shutdown, and then the serial port is reopened, the board resets itself. On the STM32 this does not occur. As a result, the flag to stop reporting is never reset on restarting the application, and therefore, the pins are never scanned again until the board is reset. I will come up with a fix to make sure the flag is cleared upon restarting the applicaton.

MrYsLab, 2020-11-15:
Resolved with release 1.3.

closed 2020-11-15 by MrYsLab