MrYsLab/telemetrix-aio #4: Digital input not giving value on init

mklomp, 2020-12-21:
When calling set_pin_mode_digital_input() the callback will only be called when the value changed. This is nice, since it keeps the data transfer low. At the moment this also means that the callback will only be called after the value changed, so I do not know the value at start (since it has not changed yet).

MrYsLab, 2020-12-21:
The initial value is zero and a message is sent on the first change and all future changes that meet the callback criteria thereafter. Could you please explain why this is an issue?

mklomp, 2020-12-22:
I have an IR sensor that gives me digital input. When I start telemetrix with this sensor pointing in open air (returning value 1) telemetrix will rightaway call the callback function and give me value 1. When, on the other hand I put my finger in front of this sensor (returning value 0) telemetrix will not call the callback function on startup, giving me no value whatsoever. So I do not know the value of the sensor. Removing my finger will change the value to 1 and trigger the callback.

Effectively, telemetrix will always call the first callback when a value is changed from 0 to 1. This is an issue since I want to know the value at start without needing the sensor to first change (from 0 to 1).

MrYsLab, 2020-12-23:
I understand what you are asking for, but I am not sure if I understand why it is necessary to receive a callback message for a non-change event. If the application at startup assumes the current sensor value is zero and does whatever startup initialization it needs to, why is a callback message necessary when there was no change in state? Please explain what value this additional message adds. Thanks.

mklomp, 2020-12-23:
I am trying to imement a getPinValue(). At the moment this can be called at runtime which then calls the set_pin_mode_digital_input(). This is not done at init because at that time I do not know what the user wants.

MrYsLab, 2020-12-23:
You are correct. Until an input pin's mode is set its state or value is essentially tri-stated. Although [buried in my documentation](https://mryslab.github.io/telemetrix/pin_modes/) I do state that "Both telemetrix and telemetrix-aio require that a pin's mode be explicitly set by calling one of the mode-setting methods before using a pin." 

If the mode is not set, then for input pins, the pin will never generate a callback, no matter what its current state is.

For output pins, the code will allow a user to output a value without first setting the pin mode, but for consistency, I imply that output modes should be set as well. I considered preventing the user from outputting data unless the pin mode was set, but this really does not add any value.

Hopefully, this sheds some light on the design.

mklomp, 2020-12-23:
I agree, but this happens after setting the pins mode. This can be seen already when running the digital pin example. My argument is that:

https://github.com/MrYsLab/Telemetrix4Arduino/blob/15a07fa747f7b74b5bcd28fd9373445a6fe48626/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1011

Should not be set to zero since that value should still be in the tri-state (at least not 0 or 1). Only after the first actual read it should be set to the actual value, which will then trigger the callback.

MrYsLab, 2020-12-23:
But physically, at startup, if nothing is connected to the pin, it is at a zero state. I could make the argument from a developer's point of view that the application should not be receiving a callback unless something physically changed on the pin. To provide configuration for both views, that is an immediate callback when the pin mode is set as an input or the opposite view, that no notification should be provided until the pin state has physically changed adds a lot of (in my opinion) unnecessary complexity. 
If the application assumes that the current value of the pin is zero before setting the pin mode to input does not seem to be unreasonable to me. 
Of course, I can be convinced otherwise if there is a compelling argument.

mklomp, 2020-12-23:
Not sure whether I complete understand your 'two views'. I don't think something extra should be implemented. I totally agree to keep things as simple as can be. The current behavior though is that it will give an immediate callback when the sensor connected to the pin returns 1 and will update on changes after that (view one), and does not give an immediate callback (only on changes, view two) when this same sensor happens to be returning 0. 

I agree that physically, if nothing is connected, the pin will return 0. The only 'problem' is that you do not know whether something is connected (probably there is, otherwise one would not ask for the value) and what the value is. So only after the arduino actually read the value from the pin, you know for sure what the value is. So, in my opinion only after the first read,

https://github.com/MrYsLab/Telemetrix4Arduino/blob/15a07fa747f7b74b5bcd28fd9373445a6fe48626/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L803-L806

one knows the actual value (which resolves the tri-state) and could store that one instead of 'unknown' (or -1, or anything not 0 or 1). The only thing that is needed is setting the initial value of last_value to eg. -1.

> If the application assumes that the current value of the pin is zero before setting the pin mode to input does not seem to be unreasonable to me.

Yes, but by doing so, the implementation is in neither view 1 nor view 2. The 'view' (and callback) depends on the startstate of the pin. To me it would make more sense to at least have an implementation of either of the views. Of which I prefer the one with callbacks due to ease of implementation (only need to init with -1), and in my opinion the reasoning that the state should only be changed from 'unknown' to either 0 or 1 after the first read.

MrYsLab, 2020-12-23:
A couple of points. The idea of a tri-state value does not reflect the actual state of the pin, it is just a logical notion. Until a digital input changes its physical state, the pin can be considered a zero. Again, if an application is assuming only to receive messages when a sensor actually changes state, it might need to add some "special" code to indicate to ignore the "non-change" report.

The second point is what to do for analog input. Since there may be a variable differential involved, a special case would need to be made to perform the read to guarantee that a callback is sent. Depending upon the differential set and the current value of the sensor, without having code to force the message an initial message might not be sent. Also, there is not a consistent maximum value reported between board types which is another complication.

I have not considered the implications for other sensors such as the HC-SR04. 

I  wish to keep things consistent between input types and not have any special cases. So if you can cope with things as they currently exist, I would prefer to not make the changes.

closed 2021-01-12 by MrYsLab