MrYsLab/telemetrix-aio #16: websockets

pmp-p, 2023-02-03:
Hi, any chances you are already adding websocket support to the WIFI server ?

i'd like to embed websocket support directly on board instead of interfacing with websockify  when running telemetrix on web with pygbag runtime ( tested on esp8266 )

MrYsLab, 2023-02-04:
@pmp-p 
I'm sorry, but I'm unclear about what you're asking. Are you asking to add websocket support to Telemetrix4Esp8266, and run Telmetrix Python code on the browser?

If that is correct, this is a very challenging task. One could modify the [get_next_command](https://github.com/MrYsLab/Telemetrix4Esp8266/blob/a90adfbdb038f1e2dbccb56d0a0bcf94bb904b36/examples/Telemetrix4Esp8266/Telemetrix4Esp8266.ino#L1466) function in Telemetrix4Esp8266, but it is not something I am willing to take on. 

If you would like to fork Telemetrix4Esp8266 and attempt it yourself, I would be happy to provide any guidance.

pmp-p, 2023-02-04:
> Are you asking to add websocket support to Telemetrix4Esp8266, and run Telmetrix Python code on the browser?
 yes, and it's already working well ( led, 4 wires steppers ... ) when using a websocket proxy. But the proxy is very annoying to get running when python is running on a mobile device. test code here but can only run on LAN without a certificate for the websocket  https://pygame-web.github.io/showroom/pythongit.html?-i#src/test_telemetrix.py

> One could modify the [get_next_command](https://github.com/MrYsLab/Telemetrix4Esp8266/blob/a90adfbdb038f1e2dbccb56d0a0bcf94bb904b36/examples/Telemetrix4Esp8266/Telemetrix4Esp8266.ino#L1466) function in Telemetrix4Esp8266

websocketing is a stream operation, i think it should happen before client.read() most likely when that buffer is filled from the socket read. In client body maybe ?

> If you would like to fork Telemetrix4Esp8266 and attempt it yourself, I would be happy to provide any guidance.

yes i'll do and thank a lot for the offer

MrYsLab, 2023-02-04:
I am closing this issue, but if you have any questions, you may enter them here, and I will still see them.

closed 2023-02-04 by MrYsLab