MrYsLab/telemetrix-aio #22: Problem with repeated reading from I2C device

pnovy, 2025-03-12:
Hi, I'm trying to control two PCF8574 modules connected to an Arduino Nano via the I2C bus (with addresses 0x20 and 0x21). There is no problem writing to both modules repeatedly, but since that doesn't return any value, I'm trying to verify the written data by reading from the module. However, this only works on the first try and I get no response on subsequent reads. The callback function is not called anymore... I ended up with only one module connected with all peripherals disconnected to isolate the problem without success. In addition to i2c_read, I've also tried i2c_read_restart_transmission, but it behaves the same.

I originally encountered the problem using Firmata Express, but the results are the same when using Telemetrix4Arduino AIO (telemetrix_aio). I'm not sure if the problem is related to the library, the I2C bus or the module itself. Any idea where the problem could be? The complete test code follows. Thanks! Pavel
<pre>
import asyncio, signal
from telemetrix_aio import telemetrix_aio
  
class App():
  board = None
  i2c_addr = 0x21
  i2c_value = None
  go_down = False
    
  def __init__(self):
    self.board = telemetrix_aio.TelemetrixAIO()
    
  def terminated(self):
    self.go_down = True

  def i2c_read_callback(self, i2c_callback_data):
    print(i2c_callback_data)
    self.i2c_value = i2c_callback_data[5]  # a single byte read from i2c

  async def mainloop(self):
# setup:
    await self.board.set_pin_mode_i2c()
    await self.board.i2c_write(self.i2c_addr, [0x0])
# test:
    await asyncio.sleep(1)
    await self.board.i2c_write(self.i2c_addr, [0xff])

    self.i2c_value = None
    await self.board.i2c_read(self.i2c_addr, None, 1, self.i2c_read_callback, 0, False)
#    await self.board.i2c_read_restart_transmission(self.i2c_addr, None, 1, self.i2c_read_callback, 0, False)
    await asyncio.sleep(.5)
    print("got %s" % self.i2c_value)  # output is "got 255"

    self.i2c_value = None
    await self.board.i2c_read(self.i2c_addr, None, 1, self.i2c_read_callback, 0, False)
#    await self.board.i2c_read_restart_transmission(self.i2c_addr, None, 1, self.i2c_read_callback, 0, False)
    await asyncio.sleep(.5)
    print("got %s" % self.i2c_value) # output is "got None"
      
    while not self.go_down:
      await asyncio.sleep(.1)
    
  async def shutdown(self):
    await self.board.i2c_write(self.i2c_addr, [0])
    await self.board.shutdown()
  
if __name__ == "__main__":
  try:
    app = App()
    if app.board is not None:
      loop = asyncio.get_event_loop()
      loop.add_signal_handler(signal.SIGTERM, app.terminated)
      loop.run_until_complete(app.mainloop())
  except KeyboardInterrupt:
    pass
  except Exception as e:
    print(e)
  finally:
    if app.board is not None:
      loop.run_until_complete(app.shutdown())
</pre>

pnovy, 2025-03-13:
Hi Alan,
I've played around with the module a bit using the Adafruit_PCF8574 library and all writes and reads work as expected. The only problem I found was that to actually make it work I need to power the module(s) via some extra data pin to ensure the module resets when the code runs (after the reset of Arduino etc.).

As a next step I modified the i2c_read() code in Telemetrix4Arduino.ino to immediately return an i2c_report_message without touching the device. I was surprised to see that the callback associated with i2c_read is still only called once. It seems to me that the problem is not related to the PCF8574...

Thanks.

MrYsLab, 2025-03-13:
Could you please post the working sketch you created using the Adafruit library?
This will help me understand precisely how Adadruit uses the wire library compared to how telemtrix uses it.

pnovy, 2025-03-14:
I'll post the sketch later today as I plan to do some more testing of the code.

I've been investigating the problem further this morning. I've modified the Telemetrix4Arduino code slightly to blink the built-in LED based on the value read from the module inside the i2c_read command routine. Since the default value of the register is 255, I first wrote value of 3 to see a reasonable result. It blinks as expected upon repeated reads from the Python script. The only problem I see is that the callback is not being called more than once for some reason...

I'm attaching a bit simplified and modified Python code (it handles the exception thrown when the Arduino is blocked by the Serial monitor, when a set_pin_mode_i2c call is missing, etc.). The PCF module is now driven by pin D2.
<pre>
import asyncio, signal
from telemetrix_aio import telemetrix_aio

class App():
  board = None
  i2c_addr = 0x21
  i2c_value = None
  go_down = False
  vcc_pin = 2

  def terminated(self):
    self.go_down = True

  def i2c_read_callback(self, i2c_callback_data):
    try:
      print(i2c_callback_data)
      self.i2c_value = i2c_callback_data[5]  # a single byte read from i2c
    except Exception as e:
      print(e)

  async def i2c_read(self, i2c_address):
    self.i2c_value = None
    await self.board.i2c_read(self.i2c_addr, None, 1, self.i2c_read_callback, 0, False)
    await asyncio.sleep(2)
    print("got %s" % self.i2c_value)

  async def mainloop(self):
    await self.board.set_pin_mode_digital_output(self.vcc_pin)
    await self.board.digital_write(self.vcc_pin, 1)
    await asyncio.sleep(1)
    await self.board.set_pin_mode_i2c()
    await self.board.i2c_write(self.i2c_addr, [3])
    while not self.go_down:
      try:
        await self.i2c_read(self.i2c_addr)
      except Exception as e:
        print(e)
      await asyncio.sleep(1)

  async def shutdown(self):
    await self.board.shutdown()

if __name__ == "__main__":
  try:
    app = App()
    app.board = telemetrix_aio.TelemetrixAIO(shutdown_on_exception = False)
    loop = asyncio.get_event_loop()
    loop.add_signal_handler(signal.SIGTERM, app.terminated)
    loop.run_until_complete(app.mainloop())
  except KeyboardInterrupt:
    pass
  except Exception as e:
    print(e)
  finally:
    if app.board is not None:
      loop.run_until_complete(app.shutdown())
</pre>

pnovy, 2025-03-14:
Just tested with two modules (expanders) connected. Since I have two 4-port mosfet modules controlled by each expander, I can see that the value written is correct (indicated by the LED per channel being lit). With two different values ​​written to both modules, i2c_read always returns the same value, which is unexpected behavior. In my own sketch, I use an instance of the Adafruit_PCF8574 class per module (and also an instance of the TwoWire class, but that's probably not necessary).

I thought why the callback isn't called is that the same value is always returned from Arduino, but that's not the case, because with one module connected the LED blinks correctly according to the value written.

pnovy, 2025-03-14:
Attaching a fully functional sketch controlling two expanders. All expander pins are used as outputs (they need to be pulled up to Vcc to properly drive the mosfets). Since I need to turn off all mosfets at startup (the expander's output defaults to 0xff, meaning everything is on), their power is controlled by another separate mosfet (VCC_PIN_MOSFET). Both expanders are powered by VCC_PIN.

My reason for using Telemetrix or Firmata Express is that I need to control multiple external things like buttons, sensors, and a NeoPixel RGB LED ring at once. It is much easier to do this from Python. I have already extended Firmata Express to control NeoPixel (I discovered Telemetrix later) via a STRING_DATA callback. At the time of writing this code, my extension for controlling the PCF85474 was unreliable and I was unable to return the read value to Python. Only two lines added to the original sketch...

<pre>
#include "Adafruit_PCF8574.h" // <> should be used instead - not shown properly in the comment
//#include "Wire.h"  // dtto

#define PCF_MAX_MODULES 2
#define PCF_CHANNEL_COUNT 8
#define VCC_PIN_MOSFET 3
#define VCC_PIN 9

class Extension {
  private:
    Adafruit_PCF8574 *pcf[PCF_MAX_MODULES];
//    TwoWire wire[PCF_MAX_MODULES];  // not necessary?!
    bool pcf_ready = false;
    int pcf_base;  // 0x00 some default?
    int pcf_modules; // 0

  public:
    Extension() {
      //for (int i = 0; i < PCF_MAX_MODULES; i++) pcf[i] = NULL;
    }

    bool pcf_init(int base, int modules) {
      if (pcf_ready || (modules > PCF_MAX_MODULES)) return false;
      bool x = true;
      for (int i = 0; i < modules; i++) {
        pcf[i] = new Adafruit_PCF8574;
        x &= pcf[i]->begin(base + i);//, &wire[i]);
        //delay(30);
        //for (int j = 0; j < PCF_CHANNEL_COUNT; j++) x &= pcf[i]->pinMode(j, OUTPUT);
        //if (!x) return x;
        x &= pcf[i]->digitalWriteByte(0x00); // all off !
        if (!x) return x;
      }
      pcf_base = base;
      pcf_modules = modules;
      pcf_ready = x; // true
      return x;
    }

    bool pcf_write(int module, int value) {
      if (!pcf_ready || (module < 0) || (module >= pcf_modules)) return false;
      Serial.println("writing " + String(value) + " to " + String(module));
      return pcf[module]->digitalWriteByte(value);
    }

    bool pcf_read(int module, int *value) {
      *value = -1;
      if (!pcf_ready || (module < 0) || (module >= pcf_modules)) return false;
      uint8_t val = pcf[module]->digitalReadByte();
      *value = (int)val;
      Serial.println("reading " + String(*value) + " from " + String(module));
      return true;
    }
};

Extension extension;

void setup() {
   pinMode(VCC_PIN, OUTPUT);
   pinMode(VCC_PIN_MOSFET, OUTPUT);
   digitalWrite(VCC_PIN, 0);
   digitalWrite(VCC_PIN_MOSFET, 0);
   delay(20);
   digitalWrite(VCC_PIN, 1);
   digitalWrite(VCC_PIN_MOSFET, 1);
   delay(200);
   Serial.begin(115200);
   while (!Serial) delay(10);
   if (extension.pcf_init(0x20, 2)) {
     Serial.println("ready!");
     Serial.println(extension.pcf_write(0, 0x32));
     Serial.println(extension.pcf_write(1, 0x07));
     int x;
     Serial.println(extension.pcf_read(0, &x));
     Serial.println(extension.pcf_read(1, &x));
    
   }
   else Serial.println("fail!");
}

void loop() {}
</pre>

pnovy, 2025-03-14:
The output of the sketch always looks like this:
<pre>
14:51:56.860 -> ready!
14:51:56.860 -> writing 50 to 0
14:51:56.860 -> 1
14:51:56.860 -> writing 7 to 1
14:51:56.860 -> 1
14:51:56.860 -> reading 50 from 0
14:51:56.860 -> 1
14:51:56.860 -> reading 7 from 1
14:51:56.860 -> 1
</pre>

MrYsLab, 2025-03-14:
Thanks for all the detailed information. Analyzing the sketch will take me a little time. I will be mapping the Adafruit library's interaction with the Arduino Wire library vs. the interaction that occurs with Wire when using Telemetrix.

When creating a Telemetrix i2c application, one typically takes an existing Arduino library, such as the Adafruit library, and a working example. Then, one maps or lists all the library calls to the Wire library and creates a Telemetrix script that interacts with the Wire library in the same way. Essentially, one needs to port the Arduino library for the device to Telemetrix.

From my work with i2c devices, when things did not work, I found that when doing the initial mapping, I did not understand something the Arduino library was doing. I would leave out essential details when translating to Telemetrix. Once I understood the difference, I got the device to work successfully. 

I want to determine whether your Python Telemetrix script exposes a bug within telemetrix-aio/Telemetrix4Arduino or if the Python script was not correctly crafted to interact appropriately with the device. The analysis will be slow because of the complexity of the Adafruit library's dependencies on other libraries.

MrYsLab, 2025-03-16:
After looking at the Adafruit examples and the code you supplied, I would like to try the simple sketch below. It establishes two pins on the PC8574 as input and two as output. It then reads and writes the pins continuously. My main interest is to ensure that i2c reads and writes succeed, not so much in the values being set or returned.

Please connect a single PCF8574 to the Arduino without using the MOSFETs. I want things to be as simple as possible.

The sketch compiles, so you should not have to deal with syntax errors. 

Please run the sketch and let me know if you see a continuous stream of print statements for the reads in the serial monitor.

If the sketch works, I can then translate it to a Telemetrix application, and we can test it again to see if multiple reads succeed.

Thanks.
```
#include <Adafruit_PCF8574.h>

/* Setup a few input pins on the PCF8574.
 * Set other PCF8574 pins as outputs.
 * Write data to outputs and then read input pins
 */

Adafruit_PCF8574 pcf;
uint8_t value_read = 0xaa;

void setup() {
  while (!Serial) { delay(10); }
  Serial.begin(115200);
  Serial.println("Adafruit PCF8574 write/read test");

  if (!pcf.begin(0x21, &Wire)) {
    Serial.println("Couldn't find PCF8574");
    while (1)
      ;
  }
  // set PCF pin 0 and 7 as output pins
  pcf.pinMode(0, OUTPUT);
  pcf.pinMode(7, OUTPUT);

  // set PCF pin 1 and 6 as input pins
  pcf.pinMode(1, INPUT);
  pcf.pinMode(6, INPUT);
}

void loop() {
  // write to output pins
  pcf.digitalWrite(0, LOW);
  delay(100);
  pcf.digitalWrite(0, HIGH);
  delay(100);
  pcf.digitalWrite(7, LOW);
  delay(100);
  pcf.digitalWrite(7, HIGH);
  delay(100);

  // read from input pins
  value_read = pcf.digitalRead(1);
  Serial.print("Pin 1 value = ");
  Serial.println(value_read);
  // set value_read to some random value
  value_read = 0x55;
  delay(100);

  value_read = pcf.digitalRead(6);
  Serial.print("Pin 6 value = ");
  Serial.println(value_read);
  // set value_read to some random value
  value_read = 0xaa;
  delay(100);
}
```

pnovy, 2025-03-17:
When I dug deeper into Telemetrix code, I finally found that the problem with the callback is that it is not an asynchronous function. What a silly mistake. That is why it behaves the same with Pymata Express and Telemetrix. Maybe it would be good idea to throw some exception in such a case... Now it seems that everything works as expected.
<pre>
import asyncio, signal
from telemetrix_aio import telemetrix_aio

class App():
  board = None
  i2c_addr = 0x21
  i2c_addr2 = 0x20
  i2c_value = None
  go_down = False

  def terminated(self):
    self.go_down = True

  async def i2c_read_callback(self, i2c_callback_data):
    try:
#      print(i2c_callback_data)
      self.i2c_value = i2c_callback_data[5]  # a single byte read from i2c
    except Exception as e:
      print(e)

  async def i2c_read(self, i2c_address):
    await self.board.i2c_read(i2c_address, None, 1, self.i2c_read_callback, 0, False)
    await asyncio.sleep(.5)
    print("got %s" % self.i2c_value)

  async def i2c_write(self, i2c_address, i2c_value):
    await self.board.i2c_write(i2c_address, i2c_value)

  async def mainloop(self):
    await self.board.set_pin_mode_i2c()
    await self.i2c_write(self.i2c_addr2, [31])
    await self.i2c_write(self.i2c_addr, [62])
    await self.i2c_read(self.i2c_addr2)
    await self.i2c_read(self.i2c_addr)
    await self.i2c_write(self.i2c_addr2, [53])
    await self.i2c_write(self.i2c_addr, [75])
    await self.i2c_read(self.i2c_addr2)
    await self.i2c_read(self.i2c_addr)
#    while not self.go_down:
#      await asyncio.sleep(1)

  async def shutdown(self):
    if self.board is not None:
      await self.board.shutdown()

if __name__ == "__main__":
  try:
    app = App()
    app.board = telemetrix_aio.TelemetrixAIO(shutdown_on_exception = False)
    loop = asyncio.get_event_loop()
    loop.add_signal_handler(signal.SIGTERM, app.terminated)
    loop.run_until_complete(app.mainloop())
  except KeyboardInterrupt:
    pass
  except Exception as e:
    print(e)
  finally:
    if app.board is not None:
      loop.run_until_complete(app.shutdown())
</pre>
The output is as follows:
<pre>
got 31
got 62
got 53
got 75
</pre>

MrYsLab, 2025-03-17:
Thanks for letting me know. I am glad you solved the issue. The next time I edit the file, I will consider adding a check for the callback to be async.