MrYsLab/telemetrix-aio #23: Support for NeoPixel

pnovy, 2025-03-19:
Hi Alan,

Aren't you planning to extend telemetrix-aio/telemetrix to support NeoPixel's LED strips/rings like telemetrix-rpi-pico does
(https://mryslab.github.io/telemetrix-rpi-pico/neopixel/)? It's the only thing I'm missing to move from self-extended Pymata Express
to Telemetrix AIO...

Thank you.

MrYsLab, 2025-03-19:
Thank you for the question. Unfortunately, I cannot provide NeoPixel support. NeoPixels require very careful and precise timing, which would cause Telemetrix4Arduino to block. 

The Pico supports NePixels because the Pico processor contains two independent "side" processors called [PIO](https://tutoduino.fr/en/pio-rp2040-en/)s. These processors run independently of the main processor and provide the necessary timing without blocking the main processor.

closed 2025-03-19 by MrYsLab

pnovy, 2025-03-20:
Could you be more specific about which part of the Telemetrix architecture might be broken when dealing with NeoPixel? For my purposes, I extended the Telemetrix AIO and Telemetrix4Arduino using the Adafruit_NeoPixel library myself, just like I did with the Pymata Express, although it took me a bit more time. I haven't encountered any problems so far.

MrYsLab, 2025-03-20:
The Adafruit library blocks in the show() method, and the length of each block depends on the number of pixels being updated. 

If your port is working for your application, that is great, but I can't take the risk of missing an event, such as a digital input change. Therefore, I won't be adding NeoPixel support.

Thanks for the suggestion.