MrYsLab/telemetrix-aio #24: Not getting a correct RPM count using the AIO version of the code, while the non-AIO version of the code does return the correct values

Unsafewater, 2025-04-03:
Hello, i'm using an arduino UNO, a DRV8871 motor driver and an IR sensor for calculating the RPM of the motor from the output of the sensor. This is part of a larger program where the usage of the async functionality is needed, including moving a stepper motor while the RPM monitoring of the DC motor is happening and an active GUI using DearPyGui. There is also going to be a raspberry pi in the same system connected to the arduino, so in a worst case, i can move the RPM counting functionality to the GPIO pins of the PI, but i would rather not do that in all honesty.

I have been testing and have managed to isolate the inconsistency I was finding in RPM measurements, to the counting of the RPM with the async function. I know from previous testing (with the telemetrix version of the same code) that a 50% PWM duty cycle should be around 5500 RPM and 100% duty cycle should be around 6700 RPM. 

If i input 50% or 100% PWM in the async code, i always get around 500 rpm back. If i input 0% PWM right after, turning off the motor, the RPM count returns 500 rpm anyways, even with the motor off. It seems as if all the sensor callbacks were registered and put into a queue that isn't able to register the callbacks fast enough, and is letting them out of the queue whenever it can, even if the motor is already stopped but the code is still running.

Here is the code i used to test this:
```
# telemetrix aio pwm test
from telemetrix_aio import telemetrix_aio
import asyncio

# Constants
count=0
tempo = 5
reverse = False
runs = 2
port = 'COM4'
stop_word = "exit"

# Pins
in_1 = 5 #in1 driver motor DC
in_2 = 6 #in2 driver motor DC
sensor_pin = 4 #sensor pin
dir_pin = 2 #dir pin driver stepper
step_pin = 3 #step pin driver stepper
enable_pin = 7 #enable pin driver stepper

# Functions
async def sensor_callback(value):
    global count
    count += 1

async def rpm_counter(): 
    global count
    await asyncio.sleep(0.3)
    start_count = count
    await asyncio.sleep(tempo)
    end_count = count

    rpm = (end_count - start_count) * 30 /(4*tempo)  # Convert to RPM
    return rpm

async def board_close(myboard):
    await myboard.analog_write(in_1, 0)
    await myboard.analog_write(in_2, 0)
    await myboard.digital_write(enable_pin, 1)

    await myboard.shutdown()
    print("DC and Stepper motors stopped, Arduino disconnected.")

async def PWM_input_loop(myboard):
    global count, rpm
    while True:
        v = input(f"\nPWM duty cycle from 20 to 100 (type '{stop_word}' to stop): ").strip()

        if v.lower() == stop_word:
            await myboard.analog_write(in_1, 0)
            await myboard.analog_write(in_2, 0)
            break

        if v.isdigit() and (int(v) == 0 or int(v) in range(20, 101)):
            if int(v) == 0:
                await myboard.analog_write(in_1, 0)
                await myboard.analog_write(in_2, 0)
                print("Stopping motor.")
                for i in range(runs):
                    rpm = await rpm_counter()
                    print(f"RPM: {rpm}")

            if int(v) in range(20, 101):
                v_fl = int((float(v) / 100.0) * 255)
                print(f"PWM: {v_fl} / 255")
                count = 0
                if reverse == True:
                    await myboard.analog_write(in_1, v_fl)
                else:
                    await myboard.analog_write(in_2, v_fl)

                for i in range(runs):
                    rpm = await rpm_counter()
                    print(f"RPM: {rpm}")
        else:
            print("Invalid input! Please enter a valid number or type 'exit' to stop.")

# Board setup
board = telemetrix_aio.TelemetrixAIO(com_port=port, close_loop_on_shutdown=False)

await board.set_pin_mode_analog_output(in_1)
await asyncio.sleep(.01)
await board.set_pin_mode_analog_output(in_2)
await asyncio.sleep(.01)
await board.set_pin_mode_digital_input(sensor_pin, callback=sensor_callback)
await asyncio.sleep(.01)
motor_step = await board.set_pin_mode_stepper(interface=1, pin1=step_pin, pin2=dir_pin)
await asyncio.sleep(.01)
await board.set_pin_mode_digital_output(enable_pin)
await asyncio.sleep(.01)
await board.digital_write(enable_pin, 1) #desligar o stepper

# Running the input function
await PWM_input_loop(board)

# Closing the connection
await board_close(board)
```

And here is some of the output from running this code:

> TelemetrixAIO Version: 1.25
> Copyright (c) 2018-2023 Alan Yorinks All rights reserved.
> 
> Opening COM4 ...
> Waiting 4 seconds for the Arduino To Reset.
> Searching for correct arduino_instance_id: 1
> Correct arduino_instance_id found
> Telemetrix4AIO found and connected to COM4
> Telemetrix4Arduino Version Number: 5.4.4
> 
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop):  50
> 
> PWM: 127 / 255
> RPM: 484.5
> RPM: 486.0
> 
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop):  100
> 
> PWM: 255 / 255
> RPM: 492.0
> RPM: 502.5
> 
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop):  0
> 
> Stopping motor.
> RPM: 499.5
> RPM: 505.5
> 
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop):  0
> 
> Stopping motor.
> RPM: 537.0
> RPM: 538.5
> 
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop):  exit
> 
> DC and Stepper motors stopped, Arduino disconnected.

MrYsLab, 2025-04-03:
I placed your code into [Pycharm](https://www.jetbrains.com/pycharm/), and it identified many errors and warnings. The errors were all "await outside async function". This begins in this section of the code:
```
# Board setup
board = telemetrix_aio.TelemetrixAIO(com_port=port, close_loop_on_shutdown=False)

await board.set_pin_mode_analog_output(in_1)
await asyncio.sleep(.01)
```
You can download a free community edition of PyCharm [here](https://www.jetbrains.com/pycharm/download/download-thanks.html?platform=linux&code=PCC).

If you prefer not to use an IDE like PyCharm, you can run your code through a tool such as [pylint](https://www.pylint.org/).

I corrected the errors and warnings. Since I have no way of testing the code, I hope the code below will solve your issue.
Please let me know if it does.
```
# telemetrix aio pwm test
from telemetrix_aio import telemetrix_aio
import asyncio

# Constants
count = 0
tempo = 5
reverse = False
runs = 2
port = 'COM4'
stop_word = "exit"

rpm = 0

# Pins
in_1 = 5  # in1 driver motor DC
in_2 = 6  # in2 driver motor DC
sensor_pin = 4  # sensor pin
dir_pin = 2  # dir pin driver stepper
step_pin = 3  # step pin driver stepper
enable_pin = 7  # enable pin driver stepper

# Functions
async def sensor_callback(value):
    global count
    count += 1

async def rpm_counter():
    global count, rpm
    await asyncio.sleep(0.3)
    start_count = count
    await asyncio.sleep(tempo)
    end_count = count

    rpm = (end_count - start_count) * 30 / (4 * tempo)  # Convert to RPM
    return rpm

async def board_close(myboard):
    await myboard.analog_write(in_1, 0)
    await myboard.analog_write(in_2, 0)
    await myboard.digital_write(enable_pin, 1)

    await myboard.shutdown()
    print("DC and Stepper motors stopped, Arduino disconnected.")

async def PWM_input_loop(myboard):
    global count, rpm
    while True:
        v = input(f"\nPWM duty cycle from 20 to 100 (type '{stop_word}' to stop): ").strip()

        if v.lower() == stop_word:
            await myboard.analog_write(in_1, 0)
            await myboard.analog_write(in_2, 0)
            break

        if v.isdigit() and (int(v) == 0 or int(v) in range(20, 101)):
            if int(v) == 0:
                await myboard.analog_write(in_1, 0)
                await myboard.analog_write(in_2, 0)
                print("Stopping motor.")
                for i in range(runs):
                    rpm = await rpm_counter()
                    print(f"RPM: {rpm}")

            if int(v) in range(20, 101):
                v_fl = int((float(v) / 100.0) * 255)
                print(f"PWM: {v_fl} / 255")
                count = 0
                if reverse:
                    await myboard.analog_write(in_1, v_fl)
                else:
                    await myboard.analog_write(in_2, v_fl)

                for i in range(runs):
                    rpm = await rpm_counter()
                    print(f"RPM: {rpm}")
        else:
            print("Invalid input! Please enter a valid number or type 'exit' to stop.")

async def startup(the_board):

    await the_board.set_pin_mode_analog_output(in_1)
    await asyncio.sleep(.01)
    await the_board.set_pin_mode_analog_output(in_2)
    await asyncio.sleep(.01)
    await the_board.set_pin_mode_digital_input(sensor_pin, callback=sensor_callback)
    await asyncio.sleep(.01)
    motor_step = await the_board.set_pin_mode_stepper(interface=1, pin1=step_pin,
                                                   pin2=dir_pin)
    await asyncio.sleep(.01)
    await the_board.set_pin_mode_digital_output(enable_pin)
    await asyncio.sleep(.01)
    await the_board.digital_write(enable_pin, 1)  # desligar o stepper

    # Running the input function
    await PWM_input_loop(board)

    # Closing the connection
    await the_board_close(board)

# get the event loop
loop = asyncio.new_event_loop()
asyncio.set_event_loop(loop)

# Board setup
board = telemetrix_aio.TelemetrixAIO(com_port=port, close_loop_on_shutdown=False)
try:
    # start the main function
    loop.run_until_complete(startup(board))
except KeyboardInterrupt:
    loop.run_until_complete(board.shutdown())
    sys.exit(0)
```

Unsafewater, 2025-04-03:
Thanks for the quick reply! I have tested the code you posted and i'm getting the following error:

> TelemetrixAIO Version: 1.25
> Copyright (c) 2018-2023 Alan Yorinks All rights reserved.
> 
> Opening COM4 ...
> Waiting 4 seconds for the Arduino To Reset.
> Searching for correct arduino_instance_id: 1
> Correct arduino_instance_id found
> Telemetrix4AIO found and connected to COM4
> Telemetrix4Arduino Version Number: 5.4.4
> 
> ---------------------------------------------------------------------------
> RuntimeError                              Traceback (most recent call last)
> Cell In[12], line 113
>     110 board = telemetrix_aio.TelemetrixAIO(com_port=port, close_loop_on_shutdown=False)
>     111 try:
>     112     # start the main function
> --> 113     loop.run_until_complete(startup(board))
>     114 except KeyboardInterrupt:
>     115     loop.run_until_complete(board.shutdown())
> 
> File C:\ProgramData\anaconda3\Lib\site-packages\nest_asyncio.py:98, in _patch_loop.<locals>.run_until_complete(self, future)
>      95 if not f.done():
>      96     raise RuntimeError(
>      97         'Event loop stopped before Future completed.')
> ---> 98 return f.result()
> 
> File C:\ProgramData\anaconda3\Lib\asyncio\futures.py:202, in Future.result(self)
>     200 self.__log_traceback = False
>     201 if self._exception is not None:
> --> 202     raise self._exception.with_traceback(self._exception_tb)
>     203 return self._result
> 
> File C:\ProgramData\anaconda3\Lib\asyncio\tasks.py:314, in Task.__step_run_and_handle_result(***failed resolving arguments***)
>     310 try:
>     311     if exc is None:
>     312         # We use the `send` method directly, because coroutines
>     313         # don't have `__iter__` and `__next__` methods.
> --> 314         result = coro.send(None)
>     315     else:
>     316         result = coro.throw(exc)
> 
> Cell In[12], line 93, in startup(the_board)
>      91 await the_board.set_pin_mode_digital_input(sensor_pin, callback=sensor_callback)
>      92 await asyncio.sleep(.01)
> ---> 93 motor_step = await the_board.set_pin_mode_stepper(interface=1, pin1=step_pin, pin2=dir_pin)
>      94 await asyncio.sleep(.01)
>      95 await the_board.set_pin_mode_digital_output(enable_pin)
> 
> File C:\ProgramData\anaconda3\Lib\site-packages\telemetrix_aio\telemetrix_aio.py:1041, in TelemetrixAIO.set_pin_mode_stepper(self, interface, pin1, pin2, pin3, pin4, enable)
>    1039 if self.shutdown_on_exception:
>    1040     await self.shutdown()
> -> 1041 raise RuntimeError(f'The Stepper feature is disabled in the server.')
> 
> RuntimeError: The Stepper feature is disabled in the server.

This is very weird to me, because the stepper was working correctly with the rest of the program. I have just now double checked and uploaded once again the server from telemetrix4arduino, making sure that the stepper functionality is selected and the error still persists.

Thanks for the help with the errors as well, i'm not the most well-versed with python programming and have been having quite a bit of trouble wrapping my brain around working with asyncio. One of the things that may be a factor in this, is the version of asyncio that i'm using which is the latest available.

MrYsLab, 2025-04-03:
I don't use Anaconda since it has caused problems in the past. Perhaps try running in a Python virtual environment from the command line. I just ran the code here without any motors or sensors and got the following:
```
/home/afy/PycharmProjects/yyy/.venv/bin/python /home/afy/PycharmProjects/yyy/step2.py 
TelemetrixAIO Version: 1.25
Copyright (c) 2018-2023 Alan Yorinks All rights reserved.

Opening all potential serial ports...

Checking /dev/ttyACM0
	/dev/ttyACM0

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1
Telemetrix4AIO found and connected to /dev/ttyACM0
Telemetrix4Arduino Version Number: 5.4.4

PWM duty cycle from 20 to 100 (type 'exit' to stop): 50
PWM: 127 / 255
RPM: 0.0
RPM: 0.0

PWM duty cycle from 20 to 100 (type 'exit' to stop): 10
Invalid input! Please enter a valid number or type 'exit' to stop.

PWM duty cycle from 20 to 100 (type 'exit' to stop): 25
PWM: 63 / 255
RPM: 0.0
RPM: 0.0

PWM duty cycle from 20 to 100 (type 'exit' to stop): 
```

Unsafewater, 2025-04-03:
Okay, I installed pycharm community and the packages i needed and for some reason jupyterlab was giving the error with your code and in pycharm it doesn't. No clue why.

Now that i ran your code, the problem i had initially still persists. The output from the sensor callback Here is the output once again:

> C:\ProgramData\anaconda3\python.exe C:\Users\User\PycharmProjects\PythonProject\test.py 
> TelemetrixAIO Version: 1.25
> Copyright (c) 2018-2023 Alan Yorinks All rights reserved.
> 
> Opening COM4 ...
> Waiting 4 seconds for the Arduino To Reset.
> Searching for correct arduino_instance_id: 1
> Correct arduino_instance_id found
> Telemetrix4AIO found and connected to COM4
> Telemetrix4Arduino Version Number: 5.4.4
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): 50
> PWM: 127 / 255
> RPM: 475.5
> RPM: 472.5
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): 100
> PWM: 255 / 255
> RPM: 474.0
> RPM: 471.0
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): 0
> Stopping motor.
> RPM: 474.0
> RPM: 474.0
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): 0
> Stopping motor.
> RPM: 472.5
> RPM: 475.5
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): exit
> DC and Stepper motors stopped, Arduino disconnected.
> Task was destroyed but it is pending!
> task: <Task pending name='Task-2' coro=<TelemetrixAIO._arduino_report_dispatcher() running at C:\ProgramData\anaconda3\Lib\site-packages\telemetrix_aio\telemetrix_aio.py:2198> wait_for=<Future pending cb=[Task.__wakeup()]>>
> 
> Process finished with exit code 0

After this, I added this 

```
     if count % 100 == 0:
         print(count, value)
```
to the sensor callback function (on your code) to test and these are the outputs now:

> 
> C:\ProgramData\anaconda3\python.exe C:\Users\User\PycharmProjects\PythonProject\test.py 
> TelemetrixAIO Version: 1.25
> Copyright (c) 2018-2023 Alan Yorinks All rights reserved.
> 
> Opening COM4 ...
> Waiting 4 seconds for the Arduino To Reset.
> Searching for correct arduino_instance_id: 1
> Correct arduino_instance_id found
> Telemetrix4AIO found and connected to COM4
> Telemetrix4Arduino Version Number: 5.4.4
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): 50
> PWM: 127 / 255
> 100 [2, 4, 0, 1743708469.9478145]
> 200 [2, 4, 0, 1743708471.4985502]
> 300 [2, 4, 0, 1743708473.0547886]
> RPM: 483.0
> 400 [2, 4, 0, 1743708474.6082373]
> 500 [2, 4, 0, 1743708476.1652336]
> 600 [2, 4, 0, 1743708477.722369]
> RPM: 483.0
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): 100
> PWM: 255 / 255
> 100 [2, 4, 0, 1743708499.9591613]
> 200 [2, 4, 0, 1743708501.5201585]
> 300 [2, 4, 0, 1743708503.074732]
> RPM: 481.5
> 400 [2, 4, 0, 1743708504.6355126]
> 500 [2, 4, 0, 1743708506.1932545]
> 600 [2, 4, 0, 1743708507.7536047]
> RPM: 480.0
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): 0
> Stopping motor.
> 700 [2, 4, 0, 1743708513.7102416]
> 800 [2, 4, 0, 1743708515.2645035]
> 900 [2, 4, 0, 1743708516.82166]
> 1000 [2, 4, 0, 1743708518.3788064]
> RPM: 481.5
> 1100 [2, 4, 0, 1743708519.9284484]
> 1200 [2, 4, 0, 1743708521.486757]
> 1300 [2, 4, 0, 1743708523.0380414]
> RPM: 481.5
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): 0
> Stopping motor.
> 1400 [2, 4, 0, 1743708527.442277]
> 1500 [2, 4, 0, 1743708529.0027778]
> 1600 [2, 4, 0, 1743708530.5651567]
> 1700 [2, 4, 0, 1743708532.1282518]
> RPM: 480.0
> 1800 [2, 4, 0, 1743708533.6817193]
> 1900 [2, 4, 0, 1743708535.2400649]
> 2000 [2, 4, 0, 1743708536.8004525]
> RPM: 483.0
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): 0
> Stopping motor.
> 2100 [2, 4, 0, 1743708540.4131796]
> 2200 [2, 4, 0, 1743708541.9729538]
> 2300 [2, 4, 0, 1743708543.5339615]
> RPM: 480.0
> 2400 [2, 4, 0, 1743708545.0984027]
> 2500 [2, 4, 0, 1743708546.6590612]
> 2600 [2, 4, 0, 1743708548.2084613]
> 2700 [2, 4, 0, 1743708549.7635956]
> RPM: 481.5
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): 0
> Stopping motor.
> 2800 [2, 4, 0, 1743708553.6756554]
> 2900 [2, 4, 0, 1743708555.2327383]
> 3000 [2, 4, 0, 1743708556.7810607]
> RPM: 481.5
> 3100 [2, 4, 0, 1743708558.3387315]
> 3200 [2, 4, 0, 1743708559.8868864]
> 3300 [2, 4, 0, 1743708561.444807]
> 3400 [2, 4, 0, 1743708562.9993956]
> RPM: 481.5
> 
> PWM duty cycle from 20 to 100 (type 'exit' to stop): exit
> DC and Stepper motors stopped, Arduino disconnected.
> Task was destroyed but it is pending!
> task: <Task pending name='Task-2' coro=<TelemetrixAIO._arduino_report_dispatcher() running at C:\ProgramData\anaconda3\Lib\site-packages\telemetrix_aio\telemetrix_aio.py:2198> wait_for=<Future pending cb=[Task.__wakeup()]>>
> 
> Process finished with exit code 0
>

MrYsLab, 2025-04-03:
The callback for the input sensor will only be called if the state of the input pin changes. Is the sensor causing a toggling of that pin? If not, the callback will never (or perhaps only once) be called.

So, if the sensor's initial state is zero, it must change to one to receive the first callback call. Then, it must return to zero for the next callback to be received.

MrYsLab, 2025-04-03:
Could you put a print statement in sensor_callback to print the current count and see if it changes?

MrYsLab, 2025-04-03:
I would suggest that you create the simplest possible example. Perhaps one that would:
1. Initialize the digital input pin for the sensor using set_pin_mode_digital_input. Have the callback print the count.
2. Initialize the digital output pins for the DC motor.
3. Start the motor and run it for a while (perhaps 30 seconds), then stop it.
4. Do not have any other code in this application - no calculations or setting up the stepper.

If this doesn't work, what is the final value the callback printed?
If you double the time to run the motor, do things fail with the same count value?
By adjusting the motor run time, is there any run-time value that succeeds?

Unsafewater, 2025-04-04:
Okay, so I did some back of the napkin calculations, and considering the rpm value i have from the non-async telemetrix version of the code, it's expected to get approx. 1470 counts / second for 5500 rpm (50% pwm). Running this code for 2 seconds, would then result in around 3000 counts total. The values being registered for the callbacks even when the motor wasn't running were around 470 rpm, aka registering around 63 counts per second. Using the 3000 counts per second and the registering speed of 63 counts per second, if my theory that the callbacks are being put into a buffer is correct, it should take around a minute for the counting outputs to stop with these parameters.

I have also followed your instructions and edited the code to make the most simple example i can, using only the in_1 pin. I'm gonna use a small time first and then double it and get an answer to your questions. I had to edit this code on my phone, so just in case i made any mistake that i didn't see, here is that code:

```
# telemetrix aio pwm test
from telemetrix_aio import telemetrix_aio
import asyncio
import time

# Constants
count = 0
tempo = 2
port = 'COM4'
rpm = 0

# Pins
in_1 = 5  # in1 driver motor DC
sensor_pin = 4  # sensor pin

# Functions
async def sensor_callback(value):
    global count
    count += 1
	date = time.strftime('%H:%M:%S', time.localtime(data[3]))
    print(count, value[2], date)

async def rpm_counter():
    global count, rpm
    await asyncio.sleep(0.3)
    start_count = count
    await asyncio.sleep(tempo)
    end_count = count

    rpm = (end_count - start_count) * 30 / (4 * tempo)  # Convert to RPM
    return rpm

async def board_close(myboard):
    await myboard.analog_write(in_1, 0)

    await myboard.shutdown()
    print("DC and Stepper motors stopped, Arduino disconnected.")

async def startup(the_board):
    await the_board.set_pin_mode_analog_output(in_1)
    await asyncio.sleep(.01)
    await the_board.set_pin_mode_digital_input(sensor_pin, callback=sensor_callback)
    await asyncio.sleep(.01)
    
    await the_board.analog_write(in_1, 127) #50% of pwm
    await asyncio.sleep(tempo)
    await await the_board.analog_write(in_1, 0)
	
	while True:
		await asyncio.sleep(tempo) #just keeping the program open while the prints are being outputted

    # Closing the connection
    await board_close(board)

# get the event loop
loop = asyncio.new_event_loop()
asyncio.set_event_loop(loop)

# Board setup
board = telemetrix_aio.TelemetrixAIO(com_port=port, close_loop_on_shutdown=False)
try:
    # start the main function
    loop.run_until_complete(startup(board))
except KeyboardInterrupt:
    loop.run_until_complete(board.shutdown())
    sys.exit(0)
```

I'm still on the way to the university, but i'll test this as soon as i get there and i'll get back to you.

Unsafewater, 2025-04-04:
I fixed the problems there were in that code and ran it. 

```
# telemetrix aio pwm test
from telemetrix_aio import telemetrix_aio
import asyncio
import time
import sys

# Constants
count = 0
tempo = 4
port = 'COM4'
rpm = 0

# Pins
in_1 = 5  # in1 driver motor DC
in_2 = 6 #in2 driver motor DC
sensor_pin = 4  # sensor pin

# Functions
async def sensor_callback(value):
    global count
    count += 1
    date = time.strftime('%H:%M:%S', time.localtime(value[3]))
    print(count, value[2], date)

async def rpm_counter():
    global count, rpm
    await board.analog_write(in_1, 127)  # 50% of pwm
    await asyncio.sleep(0.3)
    start_count = count
    await asyncio.sleep(tempo)
    end_count = count

    rpm = (end_count - start_count) * 30 / (4 * tempo)  # Convert to RPM

    return rpm

async def board_close(myboard):
    await myboard.analog_write(in_1, 0)
    await myboard.analog_write(in_2, 0)

    await myboard.shutdown()
    print("DC and Stepper motors stopped, Arduino disconnected.")

async def startup(the_board):
    await the_board.set_pin_mode_analog_output(in_1)
    await asyncio.sleep(.01)
    await the_board.set_pin_mode_analog_output(in_2)
    await asyncio.sleep(.01)
    await the_board.set_pin_mode_digital_input(sensor_pin, callback=sensor_callback)
    await asyncio.sleep(.01)
    await rpm_counter()
    print(rpm)
    await the_board.analog_write(in_2, 0)
    # await the_board.analog_write(in_1, 127)  # 50% of pwm
    await asyncio.sleep(tempo)
    await the_board.analog_write(in_1, 0)

    while True:
        await asyncio.sleep(tempo)  # just keeping the program open while the prints are being outputted

    # # Closing the connection
    # await board_close(board)

# get the event loop
loop = asyncio.new_event_loop()
asyncio.set_event_loop(loop)

# Board setup
board = telemetrix_aio.TelemetrixAIO(com_port=port, close_loop_on_shutdown=False)
try:
    # start the main function
    loop.run_until_complete(startup(board))
except KeyboardInterrupt:
    loop.run_until_complete(board_close(board))
    sys.exit(0)
```

I left the code running with motor stopped for a while, and there were no more callbacks being registered once the outputs stopped. This is the slightly truncated output, for 2 seconds of run time:

> C:\ProgramData\anaconda3\python.exe C:\Users\User\PycharmProjects\PythonProject\test-v2.py 
> TelemetrixAIO Version: 1.25
> Copyright (c) 2018-2023 Alan Yorinks All rights reserved.
> 
> Opening COM4 ...
> Waiting 4 seconds for the Arduino To Reset.
> Searching for correct arduino_instance_id: 1
> Correct arduino_instance_id found
> Telemetrix4AIO found and connected to COM4
> Telemetrix4Arduino Version Number: 5.4.4
> 1 1 13:08:19
> 2 0 13:08:19
> 3 1 13:08:19
> 4 0 13:08:19
> 5 1 13:08:19
> 6 0 13:08:19
> 7 1 13:08:19
> 8 0 13:08:19
> 9 1 13:08:19
> 10 0 13:08:19
> ...
> 44 0 13:08:19
> 45 1 13:08:19
> 46 0 13:08:19
> 47 1 13:08:20
> 48 0 13:08:20
> 49 1 13:08:20
> 50 0 13:08:20
> ...
> 107 1 13:08:20
> 108 0 13:08:20
> 109 1 13:08:20
> 110 0 13:08:20
> 111 1 13:08:21
> 112 0 13:08:21
> 113 1 13:08:21
> 114 0 13:08:21
> ...
> 144 0 13:08:21
> 145 1 13:08:21
> 146 0 13:08:21
> 476.25
> 147 1 13:08:21
> 148 0 13:08:21
> 149 1 13:08:21
> 150 0 13:08:21
> ...
> 173 1 13:08:21
> 174 0 13:08:21
> 175 1 13:08:22
> 176 0 13:08:22
> ...
> 236 0 13:08:22
> 237 1 13:08:22
> 238 0 13:08:22
> 239 1 13:08:23
> 240 0 13:08:23
> ...
> 300 0 13:08:23
> 301 1 13:08:23
> 302 0 13:08:24
> 303 1 13:08:24
> ...
> 364 0 13:08:24
> 365 1 13:08:24
> 366 0 13:08:25
> 367 1 13:08:25
> ...
> 428 0 13:08:25
> 429 1 13:08:25
> 430 0 13:08:26
> 431 1 13:08:26
> ..
> 492 0 13:08:26
> 493 1 13:08:26
> 494 0 13:08:27
> 495 1 13:08:27
> ..
> 556 0 13:08:27
> 557 1 13:08:27
> 558 0 13:08:28
> 559 1 13:08:28
> ...
> 2336 0 13:08:55
> 2337 1 13:08:55
> 2338 0 13:08:55
> 2339 1 13:08:55
> 2340 0 13:08:55
> 2341 1 13:08:56
> 2342 0 13:08:56
> 2343 1 13:08:56
> 2344 0 13:08:56
> 2345 1 13:08:56
> 2346 0 13:08:56
> 
> DC and Stepper motors stopped, Arduino disconnected.
> 
> Process finished with exit code 0

Here once again, we can see the 2346 counts over a span of 37 seconds, equaling approx. 63 counts per second as seen before.

Then i ran the code with 100% pwm, which should be around 6700 rpm and here is the, once again shortened, output:

> C:\ProgramData\anaconda3\python.exe C:\Users\User\PycharmProjects\PythonProject\test-v2.py 
> TelemetrixAIO Version: 1.25
> Copyright (c) 2018-2023 Alan Yorinks All rights reserved.
> 
> Opening COM4 ...
> Waiting 4 seconds for the Arduino To Reset.
> Searching for correct arduino_instance_id: 1
> Correct arduino_instance_id found
> Telemetrix4AIO found and connected to COM4
> Telemetrix4Arduino Version Number: 5.4.4
> 1 1 13:26:54
> 2 0 13:26:54
> 3 1 13:26:54
> 4 0 13:26:54
> 5 1 13:26:54
> 6 0 13:26:54
> 7 1 13:26:54
> ...
> 142 0 13:26:56
> 143 1 13:26:56
> 144 0 13:26:56
> 145 1 13:26:56
> 146 0 13:26:56
> 147 1 13:26:56
> 480.0
> 148 0 13:26:56
> 149 1 13:26:56
> 150 0 13:26:56
> 151 1 13:26:56
> ...
> 3568 0 13:27:50
> 3569 1 13:27:50
> 3570 0 13:27:50
> 3571 1 13:27:50
> 3572 0 13:27:50
> 3573 1 13:27:50
> 3574 0 13:27:50
> DC and Stepper motors stopped, Arduino disconnected.
> 
> Process finished with exit code 0

Once again, 63 counts/s, for 3574 counts over 56 s.

i'll do the double the time tests after lunch, and i'll update you again.

Unsafewater, 2025-04-04:
Logs for 4 seconds at 50%:

>...
> Telemetrix4AIO found and connected to COM4
> Telemetrix4Arduino Version Number: 5.4.4
> 1 1 14:48:06
> 2 0 14:48:06
> 3 1 14:48:06
> 4 0 14:48:06
> 5 1 14:48:06
> ...
> 270 0 14:48:10
> 271 1 14:48:10
> 272 0 14:48:10
> 273 1 14:48:10
> 478.125
> 274 0 14:48:10
> 275 1 14:48:10
> 276 0 14:48:10
> 277 1 14:48:10
> ...
> 4467 1 14:49:16
> 4468 0 14:49:16
> 4469 1 14:49:16
> 4470 0 14:49:16
> 4471 1 14:49:16
> DC and Stepper motors stopped, Arduino disconnected.
> 
> Process finished with exit code 0

4471 counts in 70 seconds, approx. 63 counts/s.

Logs for 4 seconds at 100% pwm:

>...
> Telemetrix4AIO found and connected to COM4
> Telemetrix4Arduino Version Number: 5.4.4
> 1 1 14:51:14
> 2 0 14:51:14
> 3 1 14:51:14
> 4 0 14:51:14
> 5 1 14:51:14
> 
> ...
> 274 0 14:51:19
> 275 1 14:51:19
> 276 0 14:51:19
> 480.0
> 277 1 14:51:19
> 278 0 14:51:19
> 279 1 14:51:19
> ...
> 6855 1 14:53:02
> 6856 0 14:53:02
> 6857 1 14:53:02
> 6858 0 14:53:02
> 6859 1 14:53:02
> 6860 0 14:53:02
> 6861 1 14:53:02
> DC and Stepper motors stopped, Arduino disconnected.
> 
> Process finished with exit code 0

6861 counts in 108 seconds, approx. 63 counts/s once again.

Does asyncio have a minimum time that it takes to assign tasks and could the callback be assigned as a new task every time it is triggered? The 63 counts/s translates roughly to a count every 15 or 16 ms.

MrYsLab, 2025-04-04:
Thanks for all the testing. At this point, I believe you are correct that the USB/Serial input buffer is getting swamped. The code that receives incoming reports is in a [task created once](https://github.com/MrYsLab/telemetrix-aio/blob/3c54dd3f86b36cb64d298c24429c4bc140e8c1ad/telemetrix_aio/telemetrix_aio.py#L339C56-L339C82), and then the code in the [task](https://github.com/MrYsLab/telemetrix-aio/blob/3c54dd3f86b36cb64d298c24429c4bc140e8c1ad/telemetrix_aio/telemetrix_aio.py#L2160) runs in a "forever" loop. This task [puts a very small delay at the bottom of the loop.
](https://github.com/MrYsLab/telemetrix-aio/blob/3c54dd3f86b36cb64d298c24429c4bc140e8c1ad/telemetrix_aio/telemetrix_aio.py#L2198)

The sleep_tune value defaults to 0.0001 seconds. To change from the default, you may try instantiating TelemetrixAIO specifying sleep_tune:

```

board = telemetrix_aio.TelemetrixAIO(com_port=port, sleep_tune = .001, close_loop_on_shutdown=False)

```

I randomly picked .001. This should help throttle the number of replies coming back. 

Could you please experiment with this value and let me know if things improve or, better yet, are solved?

Unsafewater, 2025-04-04:
> Thanks for all the testing. At this point, I believe you are correct that the USB/Serial input buffer is getting swamped. The code that receives incoming reports is in a [task created [...]

Sadly, i got the exact same behaviour. 2305 counts in 36 seconds, with a measured rpm of 480 after 2 seconds, just like before. I tried doing the sleep tune at .0001 as well and got the same results.

MrYsLab, 2025-04-04:
At this point, I am at a loss to understand the cause. 

Could you please post the code that works for the synchronous version of Telemetrix? I want to compare it to the asyncio code.

Also, what sensor are you using to detect motor motion? How does the sensor get triggered? The sensor is unlikely to be the cause,, but I am looking for any possible clues.

MrYsLab, 2025-04-04:
Also, please try adjusting sleep_tune to a very low number, such as 0.1. Do you see any change with this long delay?

MrYsLab, 2025-04-04:
I just looked more closely at the results you posted earlier today. I am unsure how to interpret the output.

Does the output showing the digital input pin toggling stop before the end of the program, or does the toggling output continue no matter how long you run the motors? The output shows that the count continues to increase as long as things are running.

Unsafewater, 2025-04-04:
So small update, I found a small error in the code where the motor was running for twice as long as it was supposed to, and fixed that. With that fix the final count was halved and is now consistent with the non async version, but the async version is still printing at the 63 counts per second rate. I ran this same code without asyncio and using the normal telemetrix and i got a similar behaviour as well, just much much less apparent. Which means the first RPM testing I made initially may have also been slightly off. 

I have increased the number of blades passing through the sensor since then as well, as I was trying to reduce the error in the calculation, which may be a factor. The RPM calculation is done as the (rpm * 60) / (tempo * num of blades * 2 counts per pass) with rpm being the returned value from the rpm counter. The sensor is a TCRT5000 on a breakout board and supposedly has a max response time of 1 ms, so i don't think that's it.

The final count in this version was 1237 with the motor running 2 seconds with the rpm output as 3982.5 RPM. Using the formula, the RPM value should have been somewhere around 4640 RPM. The code i used was this:

```
# telemetrix aio pwm test
from telemetrix import telemetrix
import time

# Constants
count = 0
tempo = 2
port = 'COM4'
rpm = 0

# Pins
in_1 = 5  # in1 driver motor DC
in_2 = 6 #in2 driver motor DC
sensor_pin = 4  # sensor pin

# Functions
def sensor_callback(value):
    global count
    count += 1
    # date = time.strftime('%H:%M:%S', time.localtime(value[3]))
    # print(count, value[2], date)

def rpm_counter():
    global count, rpm
    board.analog_write(in_1, 127)  # 50% of pwm
    time.sleep(0.3) #allow motor to accelerate
    start_count = count
    time.sleep(tempo)
    end_count = count
    board.analog_write(in_1, 0)  # 50% of pwm
    rpm = (end_count - start_count) * 60 / (2 * 4 * tempo)  # Convert to RPM

    return rpm

def board_close(my_board):
    my_board.analog_write(in_1, 0)
    my_board.analog_write(in_2, 0)

    my_board.shutdown()
    print("DC and Stepper motors stopped, Arduino disconnected.")

def startup(the_board):
    the_board.set_pin_mode_analog_output(in_1)
    time.sleep(.01)
    the_board.set_pin_mode_analog_output(in_2)
    time.sleep(.01)
    the_board.set_pin_mode_digital_input(sensor_pin, callback=sensor_callback)
    time.sleep(.01)
    rpm_counter()
    print(f"RPM: {rpm}")
    time.sleep(3)
    print(f"Final count: {count}")

# Board setup
board = telemetrix.Telemetrix(com_port=port)

startup(board)

board_close(board)
```

> Also, please try adjusting sleep_tune to a very low number, such as 0.1. Do you see any change with this long delay?

I tried this on the async version with .1 and i got back a total of 22 counts after the 2 seconds. It took like 5 minutes until it finished printing all of the counts, but the total was consistent with previous testing.

MrYsLab, 2025-04-04:
I am still confused. It looks like you are not missing any information from the sensor; it just takes a long time for all the changes to be received and processed. Is that correct?

There is transport overhead since data transfer is throttled at 115200. The data may not get across the link fast enough, and adding more blades would make things worse. With each message coming from the server, more than one transition is being detected and queued up for transmission.

Would it be possible to lessen the transitions detected by using the least possible number of blades?

Unsafewater, 2025-04-04:
I just tried running the RPM counter for 30 seconds with PWM value at 255 and giving it a minute until the code outputs the final count, to see if the callbacks are also getting buffered with the sync version. The final RPM was 6023.5 and the final count was 24405. With this final count, the calculated RPM is 6100.25. So slightly different, but much closer than with the async version.

Unsafewater, 2025-04-04:
> I am still confused. It looks like you are not missing any information from the sensor; it just takes a long time for all the changes to be received and processed. Is that correct?

Yes, that is correct.

> There is transport overhead since data transfer is throttled at 115200. The data may not get across the link fast enough, and adding more blades would make things worse. With each message coming from the server, more than one transition is being detected and queued up for transmission.

Does telemetrix-AIO use a different baudrate than the normal telemetrix? Could that be it? 
I also just checked, and the telemetrix sleeptune has a default of 0.000001 and AIO has a default of 0.001. I tried the async code with the sleeptune with the 5 zeros but still same 63 counts per second rate.

> Would it be possible to lessen the transitions detected by using the least possible number of blades?

Yeah, i can try that, but it seems to me that isn't the problem, since the normal one is being much better. Sorry if i'm being confusing anywhere in this, i'm trying to explain things as best as i can

Unsafewater, 2025-04-04:
Cutting back down to one blade would result in a max rpm of 480 * 4 = 2000 ish, with a much larger error and still with the 63 counts/second problem

MrYsLab, 2025-04-04:
The transport baud rate is 115200 for both async and sync. If you are using only a single digital input for the RPM sensor, then generating a report message for every 10 (or some other value) sensor changes might improve things.

What I am suggesting is to modify [Telemetrix4Arduino ](https://github.com/MrYsLab/Telemetrix4Arduino/blob/4e63d385d2fc87c490f141fb363b79c84fe04c6e/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1760)to accumulate sensor changes and only sending a report when the target number of sensor changes is achieved.

Other than that, I think your suggestion of using the Raspberry PI might be the only alternative.

Unsafewater, 2025-04-04:
> The transport baud rate is 115200 for both async and sync. If you are using only a single digital input for the RPM sensor, then generating a report message for every 10 (or some other value) sensor [...]

okay, could you please tell me what i need to change in the arduino code? I'd like to try it with 20 sensor changes to give some headroom for larger RPMs. The only question I have is that is if it will affect the stepper callback function, since i am also using the arduino to control the stepper while the motor speed is (ideally) being controlled by a PID controller. And thanks for all the help, i genuinely wouldn't have gotten any closer to a possible resolution without your help.

Unsafewater, 2025-04-04:
One other thing that i thought of that might be completely irrelevant is that the 63 counts/s value is almost exactly equal to 6 bits, but that does nothing in explaining the difference between sync and async codes, i guess. You talked about code overhead and tuning the sleep, but tuning the sleep could only slow down the printing of the registered sensor values, and not speed up. This could be possibly be related to asyncio's own overhead as well. I'm sorry if this makes no sense and for the spam of posts in the topic, i'm trying to brainstorm possibilities of the source of this. 

For another test, i also tried disabling every other feature define in the arduino server for another try and nothing changed.

Unsafewater, 2025-04-04:
I think i may have found something else. When running both codes back to back, it seems as if the prints are outputted in blocks of different sizes at the same speed. The async output blocks are much smaller than the sync output blocks.  

https://youtu.be/wEh1pJIKcj0 here, I hope this is able to show what i mean with the whole different sized blocks thing. I also noticed noise whenever a block of prints was output, as in interference noise on my earbuds, but i have no idea if it's audible on the video. I'm only going to be able to do more with this on monday, be it more tests or changing the code.

Unsafewater, 2025-04-04:
> I will modify the server and get back to you as soon as possible—it might not be until tomorrow. Believe it or not, asyncio is much better now than when it first appeared as a feature. However, its inner workings are somewhat mysterious and magical.
> 
> The stepper motor support won't be affected. I plan to modify the code only to affect the sensor pin, meaning you can have other digital inputs that would work "normally".
> 
> I will create a branch in the Telemetrix4Arduino repository that you can download as a .zip file. Instructions may be found [here](https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries/). The process is the same for both versions 1 and 2 of the IDE. Just scroll down the page and look for "Importing a .zip Library."

Thank you so much! I'm very glad i decided to ask for help on this, i've been  unsuccessfully trying to debug this for quite a while on my own, I only posted this here when i was almost certain i had isolated the problem and that it really wasn't just my code being a buggy mess xD

MrYsLab, 2025-04-04:
No problem. I am glad to help and only hope this solves the problem. This will work for both asyncio and sync.

To download the .zip file, click this[ link ](https://github.com/MrYsLab/Telemetrix4Arduino/tree/buffered_digital_input)to access the GitHub branch.

Next, click the green <> Code button on the top right of the page. This will download a .zip file.

Remove the current version of Telemetrix4Arduino using the Arduino IDE library manager and then do a .zip install. Then, load Telemetrix4Arduino into your board.

The PIN is defined at [this line](https://github.com/MrYsLab/Telemetrix4Arduino/blob/68f1da05d2ff7cb9bb7af18e6a657280a982392a/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1747), and the number of state changes [is defined here](https://github.com/MrYsLab/Telemetrix4Arduino/blob/68f1da05d2ff7cb9bb7af18e6a657280a982392a/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1748).

MrYsLab, 2025-04-05:
I have created yet another branch for [Telemetrix4Arduino](https://github.com/MrYsLab/Telemetrix4Arduino/tree/buffered_digital_input_timed).

Installation instructions are the same as above.

This branch implements a timed version of reporting information for the RPM sensor.

Here is the code for scanning digital inputs. The annotation is below.

```
// scan the digital input pins for changes
void scan_digital_inputs() {

#define SENSOR_PIN 4
#define RPM_SCAN_INTERVAL 1000  // send report each interval is achieved
  byte value;

  // report message

  // byte 0 = packet length
  // byte 1 = report type
  // byte 2 = pin number
  // byte 3 = value
  byte report_message[4] = { 3, DIGITAL_REPORT, 0, 0 };

  for (int i = 0; i < MAX_DIGITAL_PINS_SUPPORTED; i++) {
    if (the_digital_pins[i].pin_mode == INPUT || the_digital_pins[i].pin_mode == INPUT_PULLUP) {
      if (the_digital_pins[i].reporting_enabled) {
        // if the value changed since last read
        value = (byte)digitalRead(the_digital_pins[i].pin_number);
        if (value != the_digital_pins[i].last_value) {
          the_digital_pins[i].last_value = value;

          // process rpm sensor
          if (i == SENSOR_PIN) {
            sensor_count++;
            rpm_current_millis = millis();
            if (rpm_current_millis - rpm_previous_millis > RPM_SCAN_INTERVAL) {
              rpm_previous_millis = rpm_current_millis;

              /***************************************************/
              // do rpm calculation here
              // place result in rpm_value - a 16 bit value
              /***************************************************/

              report_message[2] = (byte)i;
              report_message[3] = value;
              Serial.write(report_message, 4);
              sensor_count = 0;
            }
          } else {
            report_message[2] = (byte)i;
            report_message[3] = value;
            Serial.write(report_message, 4);
          }
        }
      }
    }
  }
}
```
Essentially, counts are continually accumulated for the sensor pin. The RPM will be calculated at the end of a given time period, and a report is sent to the client. The format for the report is the same for "normal" digital input pins, so you will need to qualify the report for the RPM sensor by looking at the pin number in the callback (on the Python side).

The sensor PIN is defined [here](https://github.com/MrYsLab/Telemetrix4Arduino/blob/ccb62d126249a667af44dff22b85daf248ebb390/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1749).

The report interval is set [here](https://github.com/MrYsLab/Telemetrix4Arduino/blob/ccb62d126249a667af44dff22b85daf248ebb390/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1750). The value is in milliseconds. Change it for something that makes sense to you.

The special handling for the sensor pin begins [here](https://github.com/MrYsLab/Telemetrix4Arduino/blob/ccb62d126249a667af44dff22b85daf248ebb390/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1770). 

You will need to add code to perform the actual RPM calculation  [here](https://github.com/MrYsLab/Telemetrix4Arduino/blob/ccb62d126249a667af44dff22b85daf248ebb390/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1776).

The RPM value in the report is 8-bit, so you must scale things if necessary. If need be, you can compensate on the Python side in the callback.

When the message is sent to the client, the count of sensor pulses is [reset to 0 ](https://github.com/MrYsLab/Telemetrix4Arduino/blob/ccb62d126249a667af44dff22b85daf248ebb390/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1784).

Please remember that Telemetrix4Arduino has time overhead in performing command handling and various sensor scanning. Therefore, the time it takes to accomplish the loop() function may vary for each loop iteration.

Unsafewater, 2025-04-07:
> I have created yet another branch for [Telemetrix4Arduino](https://github.com/MrYsLab/Telemetrix4Arduino/tree/buffered_digital_input_timed).
>
> Installation instructions are the same as above.
>
> This [...]

Sorry for the late reply, only had time to look at this a bit ago. I looked at the bit of code for the arduino server you made, and i made a change. The time for the measurement of the rpm is something that is variable in the GUI and changeable by the user, as are the number of blades. If I understood the code correctly, this makes it so that the calculation being within the arduino server, might be a problem if the chosen sample time or the number of blades are different from the ones that are set within the arduino code. With this in mind, I altered the code you posted so that every 50 ms a report is sent with the sensor count in that timeframe. 

This way, I can keep using the python code to make the rpm calculation, and not lose out on any of the flexibility. I believe that is the best approach for this specific problem while still being able to adjust the number of blades and timeframe in the GUI. I chose 50 ms because it's still frequent enough for the callback to work as intended, and higher than the 15ms or 16ms that asyncio was taking to output each callback. Do you foresee anything going wrong while doing it this way?

I'll get back to you in a bit about testing this out.

Unsafewater, 2025-04-07:
Update: this method worked, of summing the counts that are sent back by the arduino in the 50 ms interval with asyncio. The values are the same for both versions of the code, with or without asyncio and the measured rpm and the final count are what they are supposed to be, within a margin of error. Only possible question is if i should lower the 50ms to 49, to try and have some small time to make sure all the counts accounted for in the calculation.

Once again, thanks so much for all the help, I just need to test the concurrency of the stepper moving as the DC motor is having it's speed measured.

MrYsLab, 2025-04-07:
The best way to determine whether 50 or 49 ms is best is to experiment when the application is fully implemented.