MrYsLab/telemetrix-aio #25: Report dispatcher dies silently

pnovy, 2025-04-07:
Hi, I'm using a modified version of Telemetrix AIO / Telemetrix4Arduino (extended to control Neopixel ring) in my current project. Everything works fine for hours, but occasionally the client freezes for an unknown reason. The primary symptom is that it stops to respond to capacitive button presses. It looks like the report dispatcher loop dies silently because no callbacks are being fired anymore (upon i2c_read calls, digital pin changes). I added a periodic call to i2c_read to the main loop when it is idle to see the application is still alive (interval 30 seconds). It waits for 0.2s for the callback being triggered and in case of a problem, the application is shut down and started again by an external script. Since then everything has been working fine again. The current value read from the first PCF8574 expander is printed to console.

I noticed these unexpected events last day:
<pre>
2025-04-06 04:37:16.459568: 0
2025-04-06 04:37:47.608373: None
2025-04-06 04:37:47.667988: Read from controller failed!
2025-04-06 04:37:47.862698: Shutdown
...
2025-04-06 05:24:40.719331: 0
2025-04-06 05:25:11.905410: 1743909911.9049773
2025-04-06 05:25:42.069274: 0
...
2025-04-06 07:37:44.650340: 0
2025-04-06 07:38:15.850133: 1743917895.8499715
2025-04-06 07:38:45.923849: 0
...
2025-04-06 11:40:47.339635: 8
2025-04-06 11:41:18.415502: 1743932478.415349
2025-04-06 11:41:48.493749: 8
...
</pre>

Then I added more information to the output and also added try/except section to _arduino_report_dispatcher():
<pre>
     try:
          await asyncio.sleep(self.sleep_tune)
          await self.report_dispatch[report](packet[1:])
     except Exception as e:
          print("Report loop exception: %s" % str(e))
...
2025-04-07 02:57:05.469995: PCF read callback: [10, 0, 1, 32, 0, 0, 1743987425.4699504]
2025-04-07 02:57:05.531147: PCF read: 0
2025-04-07 02:57:36.644654: PCF read: None
2025-04-07 02:57:36.710603: Read from controller failed!
2025-04-07 02:57:36.987819: Shutdown
2025-04-07 05:46:24.083866: Terminated
2025-04-07 05:46:25.319476: Start
TelemetrixAIO Version: 1.25
Copyright (c) 2018-2023 Alan Yorinks All rights reserved.

Opening all potential serial ports...

Checking /dev/ttyUSB0
<------>/dev/ttyUSB0

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1
Telemetrix4AIO found and connected to /dev/ttyUSB0
Telemetrix4Arduino Version Number: 5.4.4
Report loop exception: 2
Report loop exception: Servo Attach For Pin 0 Failed: No Available Servos
2025-04-07 02:57:43.870749: 12V power source voltage 0V is out of valid range!
2025-04-07 02:57:43.933742: 5V power source voltage 0V is out of valid range!
2025-04-07 02:57:44.009917: Problem with power sources
# two power sources are monitored by an analog inputs 6 and 7
...
2025-04-07 09:33:48.954492: [2, 7, 0, 1744011228.95448]
Report loop exception: list index out of range
2025-04-07 09:33:49.225427: [2, 4, 0, 1744011229.2254128]
2025-04-07 09:33:49.386828: [2, 4, 1, 1744011229.3868163]
...
2025-04-07 10:56:43.961087: [2, 4, 0, 1744016203.961073]
Report loop exception: Servo Attach For Pin 0 Failed: No Available Servos
Report loop exception: 4
2025-04-07 10:56:44.347396: [2, 8, 0, 1744016204.3473823]
2025-04-07 10:56:44.854950: [2, 4, 0, 1744016204.8549347]
</pre>

It looks like the wrong data are sometimes received from the Arduino and the loop dies silently in case of any unhandled exception... Now I'm printing each packet content to the output to see more.

pnovy, 2025-04-07:
> [10, 0, 1, 32, 0, 4]
> 2025-04-07 12:45:49.592327: PCF read callback: [10, 0, 1, 32, 0, 4, 1744022749.5922909]
> 2025-04-07 12:45:49.643763: PCF read: 4
> [2, 1, 3]
> Report loop exception: 1
> [4, 0]
> Report loop exception: 4
> [2, 4, 1]
> 2025-04-07 12:46:01.960004: [2, 4, 1, 1744022761.959991]
> 2025-04-07 12:46:01.997401: cap_butt pressed

MrYsLab, 2025-04-07:
From the initial  post above:

> Since then everything has been working fine again. The current value read from the first PCF8574 expander is printed to console.

When you add more code, things fail again. It isn't easy to understand what is happening without knowing how the client and server were modified. It sounds like some race condition. There may not be enough time between read/writes to the I2C bus.

It is unclear what other sensors/actuators are active in the application besides an i2c interface to the PCF8574 and whether they contribute to the behavior seen. 

I suggest writing the most uncomplicated application involving the PCF8574 and, for input, one type of sensor. If this fails, I recommend adding some debug statements to the server.

You do this by calling [send_debug_info 
](https://github.com/MrYsLab/Telemetrix4Arduino/blob/4e63d385d2fc87c490f141fb363b79c84fe04c6e/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L693) within the server. 

As an example, I have added debug statements to digital input scanning:
```
void scan_digital_inputs() {
  byte value;

  // report message

  // byte 0 = packet length
  // byte 1 = report type
  // byte 2 = pin number
  // byte 3 = value
  byte report_message[4] = { 3, DIGITAL_REPORT, 0, 0 };

  for (int i = 0; i < MAX_DIGITAL_PINS_SUPPORTED; i++) {
    if (the_digital_pins[i].pin_mode == INPUT || the_digital_pins[i].pin_mode == INPUT_PULLUP) {
      if (the_digital_pins[i].reporting_enabled) {
        // if the value changed since last read
        value = (byte)digitalRead(the_digital_pins[i].pin_number);
        if (value != the_digital_pins[i].last_value) {
          the_digital_pins[i].last_value = value;
          report_message[2] = (byte)i;
          report_message[3] = value;

          send_debug_info(0, report_message[0]);
          send_debug_info(1, report_message[1]);
          send_debug_info(2, report_message[2]);
          send_debug_info(3, report_message[3]);

          Serial.write(report_message, 4);
        }
      }
    }
  }
}
```
When I run the [digital input example](https://github.com/MrYsLab/telemetrix-aio/blob/master/examples/digital_input.py), 
I get the following output:

```
/home/afy/PycharmProjects/telemetrix-aio/.venv3/bin/python /home/afy/PycharmProjects/telemetrix-aio/examples/digital_input.py 
TelemetrixAIO Version: 1.25
Copyright (c) 2018-2023 Alan Yorinks All rights reserved.

Opening all potential serial ports...

Checking /dev/ttyACM0
	/dev/ttyACM0

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1
Telemetrix4AIO found and connected to /dev/ttyACM0
Telemetrix4Arduino Version Number: 5.4.4
DEBUG ID: 0 Value: 3
DEBUG ID: 1 Value: 2
DEBUG ID: 2 Value: 12
DEBUG ID: 3 Value: 1
Pin: 12 Value: 1 Time Stamp: 2025-04-07 10:51:21
DEBUG ID: 0 Value: 3
DEBUG ID: 1 Value: 2
DEBUG ID: 2 Value: 12
DEBUG ID: 3 Value: 0
Pin: 12 Value: 0 Time Stamp: 2025-04-07 10:51:22
```

Using send_debug_info can help pinpoint where the problem is occurring. Of course, using send_debug_info can affect timing and either "fix" or "break" things if some sort of race condition occurs.

pnovy, 2025-04-07:
Just added two function to the code to control Neopixel, nothing else. The problem occurs ocassionally and also/sometimes when the application idles. It means there are no other i/o operations performed, just simple i2c readings per 5 seconds. As you can see in my initial post, the timestamp instead of the value read is unexpectedly printed in three cases. It looks like a malformed packet was received from the Arduino and wrong data were passed to the callback. If None is returned, it means that the callback wasn't trigerred within the timeout of .2s and the report dispatcher thread is very probably dead due to an unhandled exception thrown. 

The second most common situation when it occurs is when turning three lasers (with a short delay .07s between them) resulting into three digital reports from light sensors connected to pins 6, 7 and 8 (expecting 3 [2, 6, 0], 3 [2, 7, 0] and 3 [2, 8, 0]). A release of the button may also occur meanwhile (3 [2, 4, 0]). As you can see there is a mess in two packets received: 3 [7,0,3] and 2 [4,1]:
(3 x 7 0) (3 2 4 1). Obviously, the client missed a byte at position x with value 2 and mixed up two packets together! 
<pre>
3
[2, 4, 0]
2025-04-07 16:57:01.785093: [2, 4, 0, 1744037821.7850795]
3
[2, 6, 0]
2025-04-07 16:57:01.970071: [2, 6, 0, 1744037821.9700558]
3
[7, 0, 3]
Report loop exception: Servo Attach For Pin 0 Failed: No Available Servos
2
[4, 1]
Report loop exception: 4
3
[2, 8, 0]
2025-04-07 16:57:02.171219: [2, 8, 0, 1744037822.1712046]
3
[2, 4, 0]
2025-04-07 16:57:02.562278: [2, 4, 0, 1744037822.5622635]
3
[2, 4, 1]
2025-04-07 16:57:03.046401: [2, 4, 1, 1744037823.046388]
3
[2, 4, 0]
2025-04-07 16:57:03.455490: [2, 4, 0, 1744037823.4554777]
</pre>

pnovy, 2025-04-07:
The application is running on a live machine, so I can't experiment much with it. However, this issue is not critical to the project. I just deployed the same code to a development machine and will see if the issue occurs there.

pnovy, 2025-04-07:
The same scenario on a live machine (3 2  x 1) (3 2 4 0)  with one byte missing (4):
<pre>
...
2025-04-07 19:16:04.863603: PCF read callback: [10, 0, 1, 32, 0, 0, 1744046164.8635693]
2025-04-07 19:16:04.894961: PCF read: 0
2025-04-07 19:16:06.329413 3 [2, 1, 3]
Report loop exception: 1
2025-04-07 19:16:06.329949 2 [4, 0]
Report loop exception: 4
2025-04-07 19:16:06.733427 3 [2, 4, 1]
2025-04-07 19:16:06.733567: [2, 4, 1, 1744046166.7335572]
2025-04-07 19:16:06.767244: cap_butt pressed
2025-04-07 19:16:07.339115 3 [2, 6, 0]
2025-04-07 19:16:07.339279: [2, 6, 0, 1744046167.3392694]
2025-04-07 19:16:07.439172 3 [2, 7, 0]
2025-04-07 19:16:07.439365: [2, 7, 0, 1744046167.4393551]
2025-04-07 19:16:07.540306 3 [2, 8, 0]
2025-04-07 19:16:07.540458: [2, 8, 0, 1744046167.5404491]
2025-04-07 19:16:08.180290 3 [2, 4, 0]
2025-04-07 19:16:08.180793: [2, 4, 0, 1744046168.1807628]
...
</pre>

pnovy, 2025-04-08:
Once I found this in the log this morning:
<pre>
2025-04-08 10:58:39.695597 3 [8, 0]
Report loop exception: list index out of range
</pre>
I decided to rewrite the key function responsible for reading data from Arduino:
<pre>
  async def read2(self, size = 1):
        """
        This is an asyncio adapted version of pyserial read
        that provides non-blocking read.

        :return: Array of characters
        """
        output = []
        while size:
            await asyncio.sleep(self.sleep_tune)
            data = self.my_serial.read(size)
            if data:
                output.extend(list(data))
                size -= len(data)
       return output
</pre>
<pre>
   packet_length = (await self.serial_port.read2())[0]

   packet = await self.serial_port.read2(packet_length)
</pre>
Along with the read timeout set to 0 when initializing the serial device.

pnovy, 2025-04-10:
It's really frustrating to see that after twenty hours of trouble-free operation since replacing the cable and the problematic USB flash drive that started disconnecting, one byte was lost twice again this morning...